luke-jr/bfgminer #733: Give Me Coins

Justinjhcs1, 2017-12-18:
Using BFGminer to start on Give me coins pool servers there are several to choose from but everyone I pick then put in my Username and Pass from that pool server I get this message pool setup failed What am I dong wrong and this is Default no Special Coding I did not even create a start.BAT file.
Is there a way to check the server? Ill try a Ping tracer but what am I doing wrong...

sunk818, 2017-12-27:
More details please. What is the bfgminer command and parameters you used? This isn't really a problem with bfgminer and sounds more like a configuration issue

olividir, 2020-06-19:
I actually have the same problem.
if I do `bfgminer` and fill out `URL`, `username` and `password` the script exits just with this info: `pool setup failed` 
Doing `bfgminer -o URLtoPool -u username.workername -p password` (note not actual input info), the program says that one of the three is wrong. However, I am doing copy/paste from url, username and password that worked before. 
Not sure how I can get a better log for this.