luke-jr/bfgminer #735: Memory leak in _load_default_configs()

lcatro, 2017-12-20:
ASAN Detect :

[image]

The reason is load_config() return json_error by malloc() ,but _load_default_configs() never free this return .

[image]

Fix :

```C

static
bool _load_default_configs(const char * const filepath, void * __maybe_unused userp)
{
        bool * const found_defcfg_p = userp;
        *found_defcfg_p = true;

        char* buffer = load_config(filepath, NULL);

        if (NULL != buffer)
                free(buffer);

        // Regardless of status of loading the config file, we should continue loading other defaults
        return false;
}

```

luke-jr, 2017-12-20:
Pretty minor since it is just a small leak during startup in an error condition.

Unfortunately, fixing it isn't that simple: you can't free the other return values, and ccan opt needs to handle the return value properly as well.

lcatro, 2017-12-20:
@luke-jr Oh ,I'm forget something .parse_config() will return a buffer ,and this is a static buffer .I change this buffer alloc by malloc() .

lcatro, 2017-12-20:
@luke-jr 

this is my fix code at parse_config()

```C
                        if (err) {  //  miner.c 2904
                                /* Allow invalid values to be in configuration
                                 * file, just skipping over them provided the
                                 * JSON is still valid after that. */
                                if (fileconf) {
                                        applog(LOG_ERR, "Invalid config option %s: %s", p, err);
                                        *fileconf_load_p = -1;
                                } else {
                                        char* err_buf = malloc(200);
                                        snprintf(err_buf, sizeof(err_buf), "Parsing JSON option %s: %s",
                                                p, err);
                                        free(name);
                                        return err_buf;
                                }
                        }

```

luke-jr, 2017-12-20:
This would be easier to review as a pull request...

coderxani, 2018-01-01:
@Lcatro.
 Nice designs u have here 
your codes are neatly arrange.
i have something i would love to show you.
kindly type your email address.
thank you ( ecoder2020@gmail.com)

lcatro, 2018-01-01:
@coderxani My email is m4i1f0rt3st@sina.cn .Thanks for your share : )

coderxani, 2018-01-02:
sent..
from: ecoder2020@gmail.com
coder Xani

coderxani, 2018-01-06:
Hello

Turkishcod, 2018-11-02:
> @coderxani My email is [m4i1f0rt3st@sina.cn](mailto:m4i1f0rt3st@sina.cn) .Thanks for your share : )

Sending you an email now ..thanks