luke-jr/bfgminer #739: 'struct stratumsrv_conn' has no member named 'desired_default_share_pdiff'

JohnHamiltonWillis, 2018-01-09:
I'd like to preface this by saying that I'm not sure what I can provide to make resolving this issue easier, but I'm willing and able.

After configuring BFGMiner for cpu-mining, I run the make command while in the BFGMiner directory. It gets all the way until bfgminer-driver-stratum.o and encounters this error:

driver-stratum.c: In function 'stratumsrv_init_diff':
driver-stratum.c:686:6: error: 'struct stratumsrv_conn' has no member named 'desired_default_sharepdiff'
  conn->desired_default_share_pdiff = false;
make[2]: *** [bfgminer-driver-stratum.o] Error 1
make[2]: Leaving directory '/root/bfgminer'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/root/bfgminer'
make: *** [all] Error 2

luke-jr, 2018-01-09:
That isn't line 686, and there's clearly something wrong with your compiler or clipboard... `desired_default_sharepdiff` isn't `desired_default_share_pdiff`

JohnHamiltonWillis, 2018-01-09:
Yeah, sorry mate. I typed it up by hand as best as I could. I've just confirmed that it did indeed say "driver-stratum.c:686:6" so I'm just as confused as you.

luke-jr, 2018-01-09:
Where did you get the code?

JohnHamiltonWillis, 2018-01-09:
I cloned it via git for 4.10.x

luke-jr, 2018-01-09:
A non-applicable fix got merged into 4.10.x. Reverted it.

I do suggest 5.5.0 instead, though...

luke-jr, 2018-01-09:
Fixed with bd129574ef08cff7283391913f4f5e1af64e5cd8

closed 2018-01-09 by luke-jr

JohnHamiltonWillis, 2018-01-09:
Thanks man, appreciate your help!