luke-jr/bfgminer #742: Cflags:: Invalid argument. bfgminer-2.10.12. MinGW.

openGLnoob, 2018-01-28:
Hi Luke! I'm trying to build an old bfgminer just with GBT and OpenCL support, but w/o any addons. I follow windows instruction, but on stage "make" I get this error:

**gcc.exe: error: Cflags:: Invalid argument**

Before that I set CFLAGS to "-O2 -msse2" as required.

monkins1010, 2018-03-06:
i think its something to do with not dinign the opencl.dll???  We have been able to do this by using the MinGW reimp and dlltool utilities to create an import library "libOpenCL.a" for the "OpenCL.dll" DLL. Here are the steps we followed:

step 1: Make sure you have mingw-utils installed (for reimp and dlltool). If you don't, at a MinGW prompt, type:

mingw-get install mingw-utils
step 2: Next, type the following command:

reimp OpenCL.lib
This should yield an OpenCL.def and libOpenCL.a file. If the result is something along the lines of "invalid or corrupt import library", the alternative step 2 should work.

alternative step 2: Use dlltool together with a .def file that you can find here: http://pastebin.com/f2ac38b2f . The command becomes:

dlltool -l libOpenCL.a -d OpenCL.def -A -k
You can now link against the newly created libOpenCL.a. If you still see missing symbols, add them to the .def file and repeat the dlltool command.

luke-jr, 2018-03-06:
This is almost certainly user error.

Regardless, old versions are not supported. New ones exist for a reason...

closed 2018-03-06 by luke-jr