luke-jr/bfgminer #745: 'make' error bfgminer

goodthebest, 2018-02-13:
Any suggestions?

root@Ubuntu-1604-xenial-64-minimal ~/bfgminer # make
>( cd . && ./gen-version.sh ) >version.h.new
cmp ./version.h version.h.new && rm version.h.new || mv version.h.new ./version.h
make  all-recursive
make[1]: Entering directory '/root/bfgminer'
Making all in lib
make[2]: Entering directory '/root/bfgminer/lib'
  GEN      signal.h
  GEN      string.h
make  all-recursive
make[3]: Entering directory '/root/bfgminer/lib'
make[4]: Entering directory '/root/bfgminer/lib'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/root/bfgminer/lib'
make[3]: Leaving directory '/root/bfgminer/lib'
make[2]: Leaving directory '/root/bfgminer/lib'
Making all in ccan.bfg
make[2]: Entering directory '/root/bfgminer/ccan.bfg'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/bfgminer/ccan.bfg'
Making all in libblkmaker
make[2]: Entering directory '/root/bfgminer/libblkmaker'
  CCLD     libblkmaker-0.1.la
/usr/bin/ld: /usr/local/lib/libbase58.a(base58.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libbase58.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
Makefile:723: recipe for target 'libblkmaker-0.1.la' failed
make[2]: *** [libblkmaker-0.1.la] Error 1
make[2]: Leaving directory '/root/bfgminer/libblkmaker'
Makefile:2756: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/root/bfgminer'
Makefile:1127: recipe for target 'all' failed
make: *** [all] Error 2

luke-jr, 2018-02-13:
You seem to have a strange install of libbase58 in /usr/local/lib...

closed 2018-02-13 by luke-jr

goodthebest, 2018-02-15:
I've tried several tweaks none worked, what is the work around for this? I'm using Ubuntu 16. 04 xenial x64

luke-jr, 2018-02-15:
```
rm /usr/local/lib/libbase58.*
```
Should do it...