luke-jr/bfgminer #747: Best way to create bfgminer.exe on Windows 10 64-bit?

sunk818, 2018-02-28:
Hi, what is the best way to create bfgminer.exe on Windows 10? I didn't have instructions for cross compiling on Windows Subsystem for Linux (WSL), so I was unable to create an EXE (although Linux binary was easy).  The minGW instructions are four years old. Are they still reliable? The dependency versions have changed, so using the latest versions of dependencies are causing random errors. Should we stick to four year old instructions and try to find the correct version of the dependencies? 

Or are there cross compile instructions for WSL that will create a proper Windows .EXE?
Or newer minGW instructions under Windows... Or perhaps Visual Studio 2013/2015?

Thoughts?

monkins1010, 2018-02-28:
I can post updates in the morning. To fresh download links that have compiled with mingw.. however even with successful installation of the dependencies. I get a lot if undefined variables on compilation. I think its possible im about 90% there..

luke-jr, 2018-02-28:
https://paste.ee/p/gANmW#X7LFRBSt4OMN3QzKxDDcnIkAcvegnhtD might be helpful.

Care to turn it into proper documentation? :)

sunk818, 2018-02-28:
I get stuck whenever I indicate --host is going to be Windows on Ubuntu 16.0.4 (Windows Subsystem Linux) and I get an error about uthash-dev, although I have it installed already. I think you mentioned in another thread that it may be a path issue (EXPORT?). If I don't use --host, then it compiles fine but again it is for Ubuntu 16, not Windows 10.

May I ask how you compile bfgminer for Windows? How do we replicate the compiling environment so we can successfully build your miner?

Whenever I try to run the default 5.5.0 Windows binary, it just exits out without any details. I've used jstefanop's 5.4.2 and I'm able to mine with that.

luke-jr, 2018-02-28:
The `make-release` script in the repository is what I use. The toolchain comes from Gentoo's crossdev tool.

monkins1010, 2018-03-11:
I will update the windows build doc... got it compiling in 32bit.. mingw.. only tried opencl tho.. but i think everything else works. . Annoying interface bug needs addressing.

flyboynyc, 2020-12-13:
How can i compile the source files and turn into a .exe? Please help...

monkins1010, 2020-12-13:
> How can i compile the source files and turn into a .exe? Please help...

download visual studio 2015 . download open openssl and put it in c:\openssl\.

open the ccminer.sln in visual studio. change the top dropdowns in vs to x64 and release. press build.  you will get errors. you need to include the c:\openssl\include in your includes and the appropriate libs that are in c:\openssl\libs.

Thats the general summary of what you need to do.