luke-jr/bfgminer #750: Using work with lower difficulty than device supports

sunk818, 2018-03-05:
https://github.com/luke-jr/bfgminer/blob/adfcccdab3d196df7d79df6604f1965ee6e0afb4/miner.c
Line 10525 has the same text

Trying to mine a coin on testnet, but getting this "warning"? Is it actually a warning or will bfgminer not mine the coin even if I solve the block?

luke-jr, 2018-03-05:
The warning indicates that you may lose blocks. For example, if the network difficulty is 5, and the hardware only supports difficulty 10+, you will lose half of the blocks you might have found otherwise.

closed 2018-03-05 by luke-jr

sunk818, 2018-03-05:
Difficulty is 43u on testnet and my OpenCL Intel GPU hashes at 1 megahash per second for this algorithm. Does this mean I will never find a block on testnet? We are using `getwork` because altcoin is not segwit enabled yet - so not GBT? 

Is it as simple as lowering the difficulty in `miner.c` (one line change), or more involved than that? Sorry, I'm not a programmer, so please forgive my ignorance. 

`
[2018-03-05 10:44:01] OCL 0: Using work with lower difficulty than device supports
 [2018-03-05 10:44:01] New block: ...13b54c9108d03da5 diff 0 (313.3 )
 [2018-03-05 10:44:01] New block detected on network
 [2018-03-05 10:44:01] OCL 0: Using work with lower difficulty than device supports
`