luke-jr/bfgminer #753: Error compile  expected '{' before ';' token enum cl_kernels interface;

monkins1010, 2018-03-09:
Hi i get the error below when compiling, my configure is

./configure --disable-other-drivers --enable-opencl

compiling on mingw, 32bit, all the dependencies are installed.

I didnt install anything under the:

Video card GPU mining (free):
	  llvm 3.3+          http://llvm.org/
	  clang 3.3+         http://clang.llvm.org/
	  libclc             http://libclc.llvm.org/
	  Mesa 9.2.0+        http://www.mesa3d.org/
	
	ATi/AMD video card GPU mining (non-free):
	  AMD APP SDK        http://developer.amd.com/tools/heterogeneous-computing/amd-accelerated-parallel-processing-app-sdk/

are they needed?  is the AMD one free now???

thanks
  

CC     bfgminer-miner.o
In file included from miner.c:82:0:
driver-opencl.h:35:27: error: expected '{' before ';' token
  enum cl_kernels interface;
                           ^
In file included from c:\mingw\include\rpcdce.h:10:0,
                 from c:\mingw\include\rpc.h:40,
                 from c:\mingw\include\windows.h:64,
                 from c:\mingw\include\winsock.h:52,
                 from c:\mingw\include\winsock2.h:62,
                 from winhacks.h:4,
                 from config.h:755,
                 from miner.c:14:
driver-opencl.h:35:18: error: two or more data types in declaration specifiers
  enum cl_kernels interface;

monkins1010, 2018-03-09:
added the AMD AMD APP SDK 3.0 for 32-bit and 64-bit Windows® . and copied the bin include and lib folders from the SDK to c:\mingw  but still the error

monkins1010, 2018-03-09:
```
`make[2]:` Entering directory `/home/£ikmkmz/bfgminer'
  CC     bfgminer-miner.o
In file included from miner.c:82:0:
driver-opencl.h:35:27: error: expected '{' before ';' token
  enum cl_kernels interface;
                           ^
In file included from c:\mingw\include\rpcdce.h:10:0,
                 from c:\mingw\include\rpc.h:40,
                 from c:\mingw\include\windows.h:64,
                 from c:\mingw\include\winsock.h:52,
                 from c:\mingw\include\winsock2.h:62,
                 from winhacks.h:4,
                 from config.h:755,
                 from miner.c:14:
driver-opencl.h:35:18: error: two or more data types in declaration specifiers
  enum cl_kernels interface;
                  ^
make[2]: *** [bfgminer-miner.o] Error 1
make[2]: Leaving directory `/home/£ikmkmz/bfgminer'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/£ikmkmz/bfgminer'
make: *** [all] Error `2`
```

monkins1010, 2018-03-10:
interface must be defined somewhere else in the newest windows drivers somwhere.... so i changed interface to xinterface in all .c files that use it

luke-jr, 2018-03-10:
Can you figure out where?

macterra, 2018-12-12:
Just fyi I worked around the issue the same way that @monkins1010 did, I renamed all references to the member variable `interface` to `kinterface` to get it to build (windows mingw). Let me know if I should submit a PR for this fix.

luke-jr, 2018-12-12:
Sure, but I'd still like to identify where `interface` is getting defined...

macterra, 2018-12-13:
In windows header `combaseapi.h`...

```
#define __STRUCT__ struct
#define interface __STRUCT__
```

I think this only showed up when I disabled all other drivers in configuration...

```
./configure --enable-keccak --enable-opencl --disable-other-drivers
```