luke-jr/bfgminer #755: After couple of accepted blocks all go Rjected

steopcik, 2018-03-20:
Hello.
Recently I bought Moonlander Futurebit 2 
I try to mine a coin solo, but after 10-15 accepted shares, it starts to reject all.
Algo: Scrypt
Block size: 1mb
Diff: 0.00086142
Diff. adj.: every 720 block
Settings
`bfgminer.exe --scrypt -o localhost:54595 -u user -p x -S MLD:all --set MLD:clock=384`
EXAMPLE OF REJECTED
```
 [2018-03-20 09:16:05] Found block for pool 0!
 [2018-03-20 09:16:05] Rejected 0455c876 MLD 0  Diff 901u/861u
 [2018-03-20 09:16:06] Found block for pool 0!
 [2018-03-20 09:16:06] Rejected 0363ecb8 MLD 0  Diff 1m/861u
 [2018-03-20 09:16:07] Found block for pool 0!
 [2018-03-20 09:16:07] Rejected 00a1105f MLD 0  Diff 6m/861u
 [2018-03-20 09:16:07] Found block for pool 0!
 [2018-03-20 09:16:07] Rejected 02986037 MLD 0  Diff 1m/861u
 [2018-03-20 09:16:12] Found block for pool 0!
 [2018-03-20 09:16:12] Rejected 0236744c MLD 0  Diff 1m/861u
 [2018-03-20 09:16:14] Found block for pool 0!
 [2018-03-20 09:16:14] Rejected 00536bd2 MLD 0  Diff 11m/861u
 [2018-03-20 09:16:16] Found block for pool 0!
 [2018-03-20 09:16:16] Rejected 0230f18b MLD 0  Diff 1m/861u
```
EXAMPLE OF ACCCEPTED 
```
 [2018-03-20 09:18:32] Pool 0 http://localhost:54595 alive
 [2018-03-20 09:18:32] No suitable long-poll found for http://localhost:54595
 [2018-03-20 09:18:34] Found block for pool 0!
 [2018-03-20 09:18:34] Accepted 03d167a8 MLD 0  Diff 1m/861u
 [2018-03-20 09:18:34] New block detected on network
 [2018-03-20 09:18:34] New block detected on network
 [2018-03-20 09:18:34] Found block for pool 0!
 [2018-03-20 09:18:34] Accepted 02b0333c MLD 0  Diff 1m/861u
 [2018-03-20 09:18:34] New block detected on network
 [2018-03-20 09:18:34] New block detected on network
 [2018-03-20 09:18:36] Found block for pool 0!
 [2018-03-20 09:18:36] Rejected 00597eba MLD 0  Diff 11m/861u
```

With GPU GTX1070 and CCminer, it works fine.
 How to fix that?

sunk818, 2018-03-20:
What are you mining? Is it pos and pow? Is your block chain up to date? If you use debug=1 in your .conf file what does it say about rejects?

steopcik, 2018-03-20:
Here is log file https://pastebin.com/zZYztwjn
Mining Marinecoin
It is POW only.
Yes, it is up to date.

sunk818, 2018-03-20:
What gives you the impression that this is a coin you can mine?

sunk818, 2018-03-20:
TL;DR - your USB ASIC hashes too fast for the network difficulty.

You only need one acceptable share for solo mining. The rest of your submitted shares are rejected because it is stale now for that block height.

sunk818, 2018-03-21:
If you are solo mining a low difficulty coin, the first acceptable share mines the block. I think the rejects may be coming because you found a block, but the miners found another block that is not acceptable. Miner has to wait until the previous is accepted on the network and then you mine on the new block height. At least, that's how I interpret the log.