luke-jr/bfgminer #756: BFG miner not starting up

SaNdMaN4847, 2018-04-02:
Hello,

When I open BFGminer.exe it shows me a blank black screen with a floating dot and doesn't ask for any prompts. Is there some configuring I need to do? (running windows 10 64-bit with bfgminer version 5.5.0).
Thanks

netMaReck, 2018-04-16:
Hi. I have the same Problem with same Windows and same Bfgminer Version. Like SaNdMaN4847.
Windows has done some updates. Since that it doesnt work anymore.
 I tried different GPU Driver versions (nvidia)  but there is just the black DOS Box. 
No Errors to follow so i dont understand the problem. Cgminer works in the same System. I need Bfgminer to listen on Stratum ( i mean the --stratum-port Option)  to connect my Miners to the chain.

Mamboom, 2018-05-17:
I have same problem too , running win 7 64 bit with bfg minner 5.5.0,  the command prompt just black 
and doing noting

wudc, 2018-06-08:
Is there a solution for this issue? I ran into the same problem with Windows 10 and BFGMiner 5.5 64 bit version.

luke-jr, 2018-06-08:
Try the 32-bit version

andy910304, 2019-03-08:
i try the version 32 bit and work. Thanks

babylines, 2021-01-23:
64 bit has no error but 32 bit shows:
BFD: Dwarf Error: Could not find abbrev number 66.
Failed to init bfd from (D:\Program Files\bfgminer-5.5.0-win32\libusb-1.0.dll)
0x6b80d59b : D:\Program Files\bfgminer-5.5.0-win32\libusb-1.0.dll : usbi_get_tid
0x47395a : D:\Program Files\bfgminer-5.5.0-win32\bfgminer.exe : /ljr/Bitcoin/bfgminer/make-release-tmp/bfgminer-5.5.0-tmp/bfgminer-5.5.0/lowl-usb.c (98) : in function (usb_devinfo_scan)
0x475ba1 : D:\Program Files\bfgminer-5.5.0-win32\bfgminer.exe : /ljr/Bitcoin/bfgminer/make-release-tmp/bfgminer-5.5.0-tmp/bfgminer-5.5.0/lowlevel.c (93) : in function (lowlevel_scan)
0x42d8bb : D:\Program Files\bfgminer-5.5.0-win32\bfgminer.exe : /ljr/Bitcoin/bfgminer/make-release-tmp/bfgminer-5.5.0-tmp/bfgminer-5.5.0/miner.c (12499) : in function (drv_detect_all)
0x43022d : D:\Program Files\bfgminer-5.5.0-win32\bfgminer.exe : /ljr/Bitcoin/bfgminer/make-release-tmp/bfgminer-5.5.0-tmp/bfgminer-5.5.0/miner.c (13549) : in function (main)
0x4010fd : D:\Program Files\bfgminer-5.5.0-win32\bfgminer.exe : crt1.c
Failed to init bfd from (C:\WINDOWS\SYSTEM32\ntdll.dll)
0x77cd75f4 : C:\WINDOWS\SYSTEM32\ntdll.dll : RtlGetAppContainerNamedObjectPath
Failed to init bfd from (C:\WINDOWS\SYSTEM32\ntdll.dll)
0x77cd75c4 : C:\WINDOWS\SYSTEM32\ntdll.dll : RtlGetAppContainerNamedObjectPath

SkyportDEV, 2021-02-08:
Did you complete that?
https://github.com/luke-jr/bfgminer/blob/bfgminer/windows-build.txt