luke-jr/bfgminer #757: FutureBit Moonlander 2 support by default?

diveyez, 2018-04-04:
These are quite new, but very cool for projects. I would really like to see support for these in BFGMiner by default.
The people making builds for this have many mistakes in their way of doing things. However your work is solid Luke

sunk818, 2018-08-24:
I can solo mine scrypt coins using minerd (pooler) on where they support getblocktemplate (GBT) but not getwork. I can't use bfgminer 5.4.2 with USB ASIC FutureBit Moonlander 2 (ML2). If ML2 is supported, its my hope the above solo mine using GBT will be supported as well.

jstefanop, 2018-08-24:
Problem is there is no support for GBT on scrypt currently. Ill look into it when I have some time, but this is a pretty low priority feature since you can already solo mining with a proxy or dedicated solo pools.