luke-jr/bfgminer #761: Guide on integrating a new FPGA Miner.

jahagirdar, 2018-05-03:
I am working on developing a multi-protocol miner FPGA using Xilinx Ultrascale.
Is there some documentation on how I can use BFGMiner to manage this custom setup?
i.e. 

1. How auto detection works.
2. The functions called by the S/W to assign work to devices and to monitor progress.
3. Etc. Etc. Etc.

luke-jr, 2018-05-03:
See the HACKING file, and maybe reference one of the newer minerloop_queue drivers.

jahagirdar, 2018-05-07:
Which file + code segment should be updated to inform the build system of the new driver?

I am currently grepping for one of the driver and updating all files where it is mentioned to also include similar strings for my dummy code. but I believe that there should be a better way as some of the files might be autogenerated.

luke-jr, 2018-05-07:
Look at the most recent driver additions, which should have a single commit for the basic build system stuff.

Why would there be autogenerated code?

raiden017, 2018-05-27:
Any traction on this? I am looking for Xilinx VCU1525 support.