luke-jr/bfgminer #765: Installing & Making BFGMiner on Raspberry Pi !

KamalZ18, 2018-10-19:
Hi to Luke,

I am trying to install and Make BFG Miner on my Raspberry Pi 3 Model B;
but I get errors please could you help me out ?

at the command:
 ./autogen.sh, it displayed:
Getting submodules...  Submodule 'ccan' … knc-asic …. libbase58 …. libblkmaker
I get error messages:
configure.ac:35: warning: macro 'AM_PATH_LIBGCRYPT' not found in library
configure.ac:29: warning: AM_PATH_LIBGCRYPT missing; CLI tool will not be available
configure.ac:29: warning: AM_PATH_LIBGCRYPT missing; CLI tool will not be available
and
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.

how do add AC_CONFIG_MACRO to file configure.ac ?

I have looked at configure.ac , there is 1 line at line no 31:
AC_CONFIG_MACRO_DIR([m4])

what is m4 ?

thank you
regards
Kamal H. Zada BSc

luke-jr, 2018-10-19:
Warnings are not errors.

closed 2018-10-19 by luke-jr

KamalZ18, 2018-10-20:
Hi to Luke,
thank you very much for taking time to try to solve my problem;
the error I get is at the Make command, it displayed:
make: *** No targets specified and no makefile found.  Stop.

first I did the command:
sudo apt-get install autoconf autogen libtool uthash-dev libjansson-dev libcurl4-openssl-dev libusb-dev libncurses5-dev git-core
which it didi successfully;

next I typed: git clone https://github.com/luke-jr/bfgminer.git
to download BFG Miner from GitHub;

next I typed:
pi@raspberrypi:~ $ cd bfgminer
./autogen.sh
./configure
make

I thought the warnings I got at the autogen.sh command contributed to the make failing ?

please could help me out ?

thank you
regards
Kamal H. Zada BSc.

luke-jr, 2018-10-20:
What did configure say?

KamalZ18, 2018-10-20:
Hi to Luke,

yes Configure displayed lots of checking messages and then gave an error:
checking for libevent... no
configure: error: Package requirements (libevent >= 2.0.3) were not met:
No package 'libevent' found

please find attached the full Log (text file) of what happened:
[BFGMinerInstal1.TXT](https://github.com/luke-jr/bfgminer/files/2498351/BFGMinerInstal1.TXT)

thank you
regards
Kamal H. Zada BSc.

rscinto, 2020-10-05:
sudo apt-get install libevent-dev libncurses5-dev 

This worked for me