luke-jr/bfgminer #770: Solo mining BFGminer S9 antminer Pool 1 slow/down or URL or credentials invalid

Luddite1337, 2019-03-20:
I have spent two days now reading posts and trying different stuff to get this working and still stuck..  lol  
  I am trying to run a small stable of s9 miners for solo mining on my own full node.  I realize chances are slim to none on mining a block.   Any help would be greatly appreciated! Willing to tip! 

Here are the errors I keep getting
Pool 1 slow/down or URL or credentials invalid
 [2019-03-19 18:48:05] Pool 0 slow/down or URL or credentials invalid
 [2019-03-19 18:48:05] No servers were found that could be used to get work from.
 [2019-03-19 18:48:05] Please check the details from the list below of the servers you have input
 [2019-03-19 18:48:05] Most likely you have input the wrong URL, forgotten to add a port, or have not set up workers

Here is what I have so far but I have tried SO many different things
bfgminer -o stratum+tcp://localhost:8332 -u test -p test --stratum-port=3333 -S bitmain:auto --set btm:model=S9 --no-longpoll
pause

here is the bitcoin config
server=1
rpcuser=test
rpcpassword=test
rpcallowip=192.168.1.1
rpcport=8332
listen=1
daemon=1
gen=0

I found if I put a bitcoin.conf in the AppData/Roaming/bitcoin folder that BFGminer starts a second pool.  Tried so many things to get this working..  Thanks for any help!

2-Watts, 2019-05-23:
[logfile.txt](https://github.com/luke-jr/bfgminer/files/3213773/logfile.txt)

works:
machine 'A' (bfgminer, w/ ASIC):
$ sudo bfgminer -o http://192.168.1.2<machine B's IP>:18332 -u test -p test --coinbase-address 2Mttvmg6ZeCr8eLNCNiLnHqcnhbkSKinZYE

machine 'B' (bitcoind-0.17.0 - full node, running testnet):
$ bitcoind -conf=/home/<user>/.bitcoin/bitcoin-core-testnet.conf

where -conf=/home/<user>/.bitcoin/bitcoin-core-testnet.conf contains:

daemon=1
dbcache=14096
datadir=</path/to/non-default data dir>
debug=rpc
testnet=1
server=1
rest=1
</path/to/non-default wallet name>.dat
mintxfee=0.0000001
walletrbf=0
upgradewallet=0
walletbroadcast=0
flushwallet=0
privdb=0
[test]
rpcuser=test
rpcpassword=test
rpcbind=<machine B's IP>192.168.1.2
rpcbind=127.0.0.1
rpcport=18332
rpcallowip=127.0.0.1
rpcallowip=<machine A's IP>192.168.1.1

However, the following doesn't work:
machine 'A' (bitcoind-0.18.0 - full node, running testnet):
$ bitcoind -conf=/home/<user>/.bitcoin/bitcoin-core-testnet.conf

where -conf=/home/<user>/.bitcoin/bitcoin-core-testnet.conf contains:

daemon=1
dbcache=1024
debug=rpc
testnet=1
server=1
rest=1
mintxfee=0.0000001
walletrbf=0
upgradewallet=0
walletbroadcast=0
flushwallet=0
privdb=0
[test]
rpcuser=test
rpcpassword=test
rpcbind=<machine A's IP>192.168.1.1
rpcbind=127.0.0.1
rpcport=18332
rpcallowip=127.0.0.1
rpcallowip=<machine A's IP>192.168.1.1

and where bfgminer is launched via:
sudo bfgminer -o http://127.0.0.1:18332 --verbose -DT -u test -p test --coinbase-address 2Mttvmg6ZeCr8eLNCNiLnHqcnhbkSKinZYE > logfile.txt #see attached

in either case, bitcoind log doesn't show any obvious error; only that bfgminer connected, like:
2019-05-23T19:11:24Z ThreadRPCServer method=getblocktemplate user=test
2019-05-23T19:11:24Z ThreadRPCServer method=getwork user=test

wmcorless, 2024-03-18:
make sure you are using bitcoind 0.17.0 or earlier. Also make sure you have --generate-to <address> in your command line.