luke-jr/bfgminer #785: Error before make command (Arch Linux x86_64)

sthanduther, 2020-11-06:
Before _./configure_ command, everything ok. But, before _make_ command, i have this  error:

> /usr/bin/ld: bfgminer-lowl-spi.o:/media/lera/b1a4b0ee-bd95-4c39-a19c-5cb0e0ee5f4b/bfgminer/bfgminer-5.5.0/util.h:220: multiple definition of `cgsleep_us_r'; bfgminer-miner.o:/media/lera/b1a4b0ee-bd95-4c39-a19c-5cb0e0ee5f4b/bfgminer/bfgminer-5.5.0/./util.h:220: first defined here
> collect2: error: ld returned 1 exit status
> make[2]: *** [Makefile:1257: bfgminer] Error 1
> make[2]: Leaving directory '/run/media/lera/b1a4b0ee-bd95-4c39-a19c-5cb0e0ee5f4b/bfgminer/bfgminer-5.5.0'
> make[1]: *** [Makefile:2755: all-recursive] Error 1
> make[1]: Leaving directory '/run/media/lera/b1a4b0ee-bd95-4c39-a19c-5cb0e0ee5f4b/bfgminer/bfgminer-5.5.0'
> make: *** [Makefile:1126: all] Error 2
>

sthanduther, 2020-11-07:
Forget. Problem solved. But, i have a old problem in mining. With or without device (CPU/GPU etc), before many minutes, the BFGMiner crash with "core dumped" message.

lubosz, 2021-02-01:
Instead of writing about another issue you have you could have said how you solved the problem... I am seeing this too on arch.

lubosz, 2021-02-01:
This is how i solved it:
```
diff --git a/util.h b/util.h
index 97321215..9f59b8ba 100644
--- a/util.h
+++ b/util.h
@@ -217,7 +217,7 @@ void cgsleep_us(int64_t us);
 #define cgtimer_time(ts_start) timer_set_now(ts_start)
 #define cgsleep_prepare_r(ts_start) cgtimer_time(ts_start)
 void cgsleep_ms_r(cgtimer_t *ts_start, int ms);
-void (*cgsleep_us_r)(cgtimer_t *ts_start, int64_t us);
+extern void (*cgsleep_us_r)(cgtimer_t *ts_start, int64_t us);
 
 static inline
 int cgtimer_to_ms(cgtimer_t *cgt)
```

I'm using a fork for futurebit moonlander, i guess upstream probably fixed this.

kdiogenes, 2022-03-31:
@lubosz, after making the modification you hinted I'm still getting the following error:

```
make[3]: Leaving directory '/home/kadu/Downloads/bfgminer-5.5.0/libblkmaker'
  CCLD     bfgminer
/usr/bin/ld: bfgminer-sha256_sse4_amd64.o:/home/kadu/Downloads/bfgminer-5.5.0/sha256_sse4_amd64.c:51: multiple definition of `g_4sha256_k'; bfgminer-sha256_sse2_amd64.o:/home/kadu/Downloads/bfgminer-5.5.0/sha256_sse2_amd64.c:51: first defined here
/usr/bin/ld: x86_64/libx8664.a(sha256_xmm_amd64.o): warning: relocation against `sha256_init_sse2' in read-only section `.text'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1257: bfgminer] Error 1
make[2]: Leaving directory '/home/kadu/Downloads/bfgminer-5.5.0'
make[1]: *** [Makefile:2755: all-recursive] Error 1
make[1]: Leaving directory '/home/kadu/Downloads/bfgminer-5.5.0'
make: *** [Makefile:1126: all] Error 2
```
Do you have any idea?

kdiogenes, 2022-04-01:
I removed the `yasm` package from my linux and run `./configure --enable-cpumining` and `make` and it's worked.

kdiogenes, 2022-04-21:
I just adopted the bfgminer AUR package and made it works with CPU and GPU support: https://aur.archlinux.org/packages/bfgminer

The problem with the multiple definition errors is a change on GCC (I guess that was in version 10) defaulting to -fno-common instead of -fcommon. When running make for this project you can use `make CFLAGS="-fcommon"` to avoid the multiple definition errors.