luke-jr/bfgminer #787: Compilling in Ubuntu 20.04 (Fail)

sthanduther, 2020-11-09:
Before I downloaded the GIT repository, I followed all the steps.

> ./autogen.sh
> ./configure --enable-opencl (I have just a AMDRaedeon GPU)
> 

After the "make" command, I have this error:

> bfgminer/miner.c:11958: undefined reference to `cpus'

I don't now how I solve this.

Dextrac, 2020-11-16:
Same issue here.

**output:**
$ make
( cd . && ./gen-version.sh ) >version.h.new
cmp ./version.h version.h.new && rm version.h.new || mv version.h.new ./version.h
make  all-recursive
make[1]: Entering directory '/home/aj/bfgminer'
Making all in lib
make[2]: Entering directory '/home/aj/bfgminer/lib'
  GEN      signal.h
  GEN      string.h
make  all-recursive
make[3]: Entering directory '/home/aj/bfgminer/lib'
make[4]: Entering directory '/home/aj/bfgminer/lib'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/home/aj/bfgminer/lib'
make[3]: Leaving directory '/home/aj/bfgminer/lib'
make[2]: Leaving directory '/home/aj/bfgminer/lib'
Making all in ccan.bfg
make[2]: Entering directory '/home/aj/bfgminer/ccan.bfg'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/aj/bfgminer/ccan.bfg'
Making all in libbase58
make[2]: Entering directory '/home/aj/bfgminer/libbase58'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/aj/bfgminer/libbase58'
Making all in libblkmaker
make[2]: Entering directory '/home/aj/bfgminer/libblkmaker'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/aj/bfgminer/libblkmaker'
make[2]: Entering directory '/home/aj/bfgminer'
make -C lib
make[3]: Entering directory '/home/aj/bfgminer/lib'
make  all-recursive
make[4]: Entering directory '/home/aj/bfgminer/lib'
make[5]: Entering directory '/home/aj/bfgminer/lib'
make[5]: Nothing to be done for 'all-am'.
make[5]: Leaving directory '/home/aj/bfgminer/lib'
make[4]: Leaving directory '/home/aj/bfgminer/lib'
make[3]: Leaving directory '/home/aj/bfgminer/lib'
  CC       bfgminer-miner.o
  CC       bfgminer-deviceapi.o
  CC       bfgminer-util.o
  CC       bfgminer-sha2.o
  CC       bfgminer-api.o
( cd . && ./gen-version.sh ) >version.h.new
cmp ./version.h version.h.new && rm version.h.new || mv version.h.new ./version.h
  CC       bfgminer-logging.o
  CC       bfgminer-driver-proxy.o
  CC       bfgminer-driver-stratum.o
  CC       malgo/bfgminer-sha256d.o
  CC       bfgminer-lowl-vcom.o
  CC       bfgminer-binloader.o
  CC       bfgminer-lowlevel.o
  CC       bfgminer-dynclock.o
  CC       bfgminer-work2d.o
  CC       bfgminer-driver-bitforce.o
  CC       bfgminer-driver-bigpic.o
  CC       bfgminer-driver-drillbit.o
  CC       bfgminer-driver-twinfury.o
  CC       bfgminer-driver-icarus.o
  CC       bfgminer-driver-cairnsmore.o
  CC       bfgminer-driver-erupter.o
  CC       bfgminer-driver-antminer.o
  CC       bfgminer-gc3355.o
  CC       bfgminer-driver-dualminer.o
  CC       bfgminer-driver-avalon.o
  CC       bfgminer-driver-avalonmm.o
  CC       bfgminer-driver-modminer.o
  CC       bfgminer-driver-bifury.o
  CC       bfgminer-driver-bitfury.o
  CC       bfgminer-libbitfury.o
  CC       bfgminer-driver-littlefury.o
  CC       bfgminer-driver-hashfast.o
  CC       bfgminer-driver-rockminer.o
  CC       bfgminer-lowl-pci.o
  CC       bfgminer-lowl-spi.o
make -C libbase58
make[3]: Entering directory '/home/aj/bfgminer/libbase58'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/aj/bfgminer/libbase58'
make -C libblkmaker
make[3]: Entering directory '/home/aj/bfgminer/libblkmaker'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/aj/bfgminer/libblkmaker'
  CCLD     bfgminer.exe
/usr/lib/gcc/x86_64-pc-msys/10.2.0/../../../../x86_64-pc-msys/bin/ld: bfgminer-miner.o:miner.c:(.rdata$.refptr.cpus[.refptr.cpus]+0x0): undefined reference to `cpus'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1340: bfgminer.exe] Error 1
make[2]: Leaving directory '/home/aj/bfgminer'
make[1]: *** [Makefile:2844: all-recursive] Error 1
make[1]: Leaving directory '/home/aj/bfgminer'
make: *** [Makefile:1209: all] Error 2

Cephandrius, 2020-12-14:
I was having the same issue as well. I found the definition of cpus in driver-cpu.h. I added the --enable-cpumining and bfgminer compiled.
$ ./configure --enable-cpumining

sthanduther, 2020-12-15:
Now I tried in Arch Linux with "--enable-cpumining" and the _make_ command show this error:

> make[2]: *** No rule to make target '../ccan-upstream/ccan/opt/helpers.c', needed by '../ccan-upstream/ccan/opt/libccan_a-helpers.o'.  Stop.
> make[2]: Leaving directory '/run/media/lera/b1a4b0ee-bd95-4c39-a19c-5cb0e0ee5f4b/bfgminer/bfgminer-master/ccan.bfg'
> make[1]: *** [Makefile:2845: all-recursive] Error 1
> make[1]: Leaving directory '/run/media/lera/b1a4b0ee-bd95-4c39-a19c-5cb0e0ee5f4b/bfgminer/bfgminer-master'
> make: *** [Makefile:1210: all] Error 2
>

Cephandrius, 2020-12-16:
Do you have this file?
(bfg src)/ccan-upstream/ccan/opt/helpers.c
If not, download bfgminer again and run the setup and make commands again.
$ ./autogen.sh
$ ./configure --enable-cpumining
$ make

sthanduther, 2020-12-17:
autogen.sh return:

> 
> Updating version.h...
> cmp: version.h: No such file or directory

dragonista89, 2021-01-08:
Hi I need some help please. Was getting same error and followed instructions and got bfgminer to work but it only mines CPU. I would like to use my TTbitT miner through USB hub but bfgminer will not detect my USB ports, seems like its only on CPU mining mode. Any assistance would be greatly appreciated, thanks in advance.

RL-R, 2021-01-10:
https://github.com/luke-jr/bfgminer/issues/799#issuecomment-757537418

revdeluxe, 2021-11-08:
@luke-jr is there a way to release a pre-compiled .x86_64? or .exec...