luke-jr/bfgminer #788: Ubuntu 18.04 BfgMiner Error

EXEDC, 2020-11-10:
Hello dear Community,
ive got a Problem while trying to install BfgMiner on Ubuntu.
Heres what happens when i start "make":

**root@vmd62441:/home/bfgminer/bfgminer#** make
( cd . && ./gen-version.sh ) >version.h.new
cmp ./version.h version.h.new && rm version.h.new || mv version.h.new ./version.h
make  all-recursive
make[1]: Entering directory '/home/bfgminer/bfgminer'
Making all in lib
make[2]: Entering directory '/home/bfgminer/bfgminer/lib'
make  all-recursive
make[3]: Entering directory '/home/bfgminer/bfgminer/lib'
make[4]: Entering directory '/home/bfgminer/bfgminer/lib'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/home/bfgminer/bfgminer/lib'
make[3]: Leaving directory '/home/bfgminer/bfgminer/lib'
make[2]: Leaving directory '/home/bfgminer/bfgminer/lib'
Making all in ccan.bfg
make[2]: Entering directory '/home/bfgminer/bfgminer/ccan.bfg'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/bfgminer/bfgminer/ccan.bfg'
Making all in libbase58
make[2]: Entering directory '/home/bfgminer/bfgminer/libbase58'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/bfgminer/bfgminer/libbase58'
Making all in libblkmaker
make[2]: Entering directory '/home/bfgminer/bfgminer/libblkmaker'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/bfgminer/bfgminer/libblkmaker'
make[2]: Entering directory '/home/bfgminer/bfgminer'
make -C lib
make[3]: Entering directory '/home/bfgminer/bfgminer/lib'
make  all-recursive
make[4]: Entering directory '/home/bfgminer/bfgminer/lib'
make[5]: Entering directory '/home/bfgminer/bfgminer/lib'
make[5]: Nothing to be done for 'all-am'.
make[5]: Leaving directory '/home/bfgminer/bfgminer/lib'
make[4]: Leaving directory '/home/bfgminer/bfgminer/lib'
make[3]: Leaving directory '/home/bfgminer/bfgminer/lib'
( cd . && ./gen-version.sh ) >version.h.new
cmp ./version.h version.h.new && rm version.h.new || mv version.h.new ./version.h
make -C libbase58
make[3]: Entering directory '/home/bfgminer/bfgminer/libbase58'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/bfgminer/bfgminer/libbase58'
make -C libblkmaker
make[3]: Entering directory '/home/bfgminer/bfgminer/libblkmaker'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/bfgminer/bfgminer/libblkmaker'
  CCLD     bfgminer
/usr/bin/ld: bfgminer-miner.o: in function `_bfg_clean_up':
/home/bfgminer/bfgminer/miner.c:11958: undefined reference to `cpus'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1341: bfgminer] Error 1
make[2]: Leaving directory '/home/bfgminer/bfgminer'
make[1]: *** [Makefile:2845: all-recursive] Error 1
make[1]: Leaving directory '/home/bfgminer/bfgminer'
make: *** [Makefile:1210: all] Error 2

Dextrac, 2020-11-16:
I am currently experiencing the same issue
$ make
( cd . && ./gen-version.sh ) >version.h.new
cmp ./version.h version.h.new && rm version.h.new || mv version.h.new ./version.h
make  all-recursive
make[1]: Entering directory '/home/aj/bfgminer'
Making all in lib
make[2]: Entering directory '/home/aj/bfgminer/lib'
  GEN      signal.h
  GEN      string.h
make  all-recursive
make[3]: Entering directory '/home/aj/bfgminer/lib'
make[4]: Entering directory '/home/aj/bfgminer/lib'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/home/aj/bfgminer/lib'
make[3]: Leaving directory '/home/aj/bfgminer/lib'
make[2]: Leaving directory '/home/aj/bfgminer/lib'
Making all in ccan.bfg
make[2]: Entering directory '/home/aj/bfgminer/ccan.bfg'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/aj/bfgminer/ccan.bfg'
Making all in libbase58
make[2]: Entering directory '/home/aj/bfgminer/libbase58'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/aj/bfgminer/libbase58'
Making all in libblkmaker
make[2]: Entering directory '/home/aj/bfgminer/libblkmaker'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/aj/bfgminer/libblkmaker'
make[2]: Entering directory '/home/aj/bfgminer'
make -C lib
make[3]: Entering directory '/home/aj/bfgminer/lib'
make  all-recursive
make[4]: Entering directory '/home/aj/bfgminer/lib'
make[5]: Entering directory '/home/aj/bfgminer/lib'
make[5]: Nothing to be done for 'all-am'.
make[5]: Leaving directory '/home/aj/bfgminer/lib'
make[4]: Leaving directory '/home/aj/bfgminer/lib'
make[3]: Leaving directory '/home/aj/bfgminer/lib'
  CC       bfgminer-miner.o
  CC       bfgminer-deviceapi.o
  CC       bfgminer-util.o
  CC       bfgminer-sha2.o
  CC       bfgminer-api.o
( cd . && ./gen-version.sh ) >version.h.new
cmp ./version.h version.h.new && rm version.h.new || mv version.h.new ./version.h
  CC       bfgminer-logging.o
  CC       bfgminer-driver-proxy.o
  CC       bfgminer-driver-stratum.o
  CC       malgo/bfgminer-sha256d.o
  CC       bfgminer-lowl-vcom.o
  CC       bfgminer-binloader.o
  CC       bfgminer-lowlevel.o
  CC       bfgminer-dynclock.o
  CC       bfgminer-work2d.o
  CC       bfgminer-driver-bitforce.o
  CC       bfgminer-driver-bigpic.o
  CC       bfgminer-driver-drillbit.o
  CC       bfgminer-driver-twinfury.o
  CC       bfgminer-driver-icarus.o
  CC       bfgminer-driver-cairnsmore.o
  CC       bfgminer-driver-erupter.o
  CC       bfgminer-driver-antminer.o
  CC       bfgminer-gc3355.o
  CC       bfgminer-driver-dualminer.o
  CC       bfgminer-driver-avalon.o
  CC       bfgminer-driver-avalonmm.o
  CC       bfgminer-driver-modminer.o
  CC       bfgminer-driver-bifury.o
  CC       bfgminer-driver-bitfury.o
  CC       bfgminer-libbitfury.o
  CC       bfgminer-driver-littlefury.o
  CC       bfgminer-driver-hashfast.o
  CC       bfgminer-driver-rockminer.o
  CC       bfgminer-lowl-pci.o
  CC       bfgminer-lowl-spi.o
make -C libbase58
make[3]: Entering directory '/home/aj/bfgminer/libbase58'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/aj/bfgminer/libbase58'
make -C libblkmaker
make[3]: Entering directory '/home/aj/bfgminer/libblkmaker'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/aj/bfgminer/libblkmaker'
  CCLD     bfgminer.exe
/usr/lib/gcc/x86_64-pc-msys/10.2.0/../../../../x86_64-pc-msys/bin/ld: bfgminer-miner.o:miner.c:(.rdata$.refptr.cpus[.refptr.cpus]+0x0): undefined reference to `cpus'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1340: bfgminer.exe] Error 1
make[2]: Leaving directory '/home/aj/bfgminer'
make[1]: *** [Makefile:2844: all-recursive] Error 1
make[1]: Leaving directory '/home/aj/bfgminer'
make: *** [Makefile:1209: all] Error 2

scorninpc, 2020-11-20:
Same on Debian 10

dbass81, 2021-02-03:
Use ./configure --enable-cpumining to fix this error.