luke-jr/bfgminer #798: cannot find libtinfow library

coreyhahn, 2020-12-31:
configure and make all complete normally.  But upon execution in the build directory 

./bfgminer -o http://localhost:8332 -u user -p password

i get the error.

./bfgminer: error while loading shared libraries: libtinfow.so.6: cannot open shared object file: No such file or directory

coreyhahn, 2020-12-31:
I see i have libtinfo library installed but i see no way to install a libtinfo**w** libarary.

coreyhahn, 2021-01-01:
sudo ln -s /lib/x86_64-linux-gnu/libtic.so.6.2 /lib/x86_64-linux-gnu/libtinfow.so.6

fixed me up for now.

closed 2021-01-01 by coreyhahn

clemente0420, 2022-03-02:
> sudo ln -s /lib/x86_64-linux-gnu/libtic.so.6.2 /lib/x86_64-linux-gnu/libtinfow.so.6
> 
> fixed me up for now.

that help me , thx