luke-jr/bfgminer #799: Compiling Centos 7

Mig75seo, 2021-01-03:
Hello,
I'm trying to compile but I have the following error when I execute the make command:

 CC       bfgminer-driver-hashfast.o
  CC       bfgminer-driver-rockminer.o
  CC       bfgminer-lowl-pci.o
  CC       bfgminer-lowl-spi.o
make -C libbase58
make[3]: Entering directory `/bitcoin/mining/bfgminer/libbase58'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/bitcoin/mining/bfgminer/libbase58'
make -C libblkmaker
make[3]: Entering directory `/bitcoin/mining/bfgminer/libblkmaker'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/bitcoin/mining/bfgminer/libblkmaker'
  CCLD     bfgminer
bfgminer-miner.o: In function `_bfg_clean_up':
/bitcoin/mining/bfgminer/miner.c:11958: undefined reference to `cpus'
collect2: error: ld returned 1 exit status
make[2]: *** [bfgminer] Error 1
make[2]: Leaving directory `/bitcoin/mining/bfgminer'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/bitcoin/mining/bfgminer'
make: *** [all] Error 2

Any idea ?

Thank you in advance.

Miguel

Flossed, 2021-01-07:
Hi miguel , 
issue occures when you compile without the CPU driver to fix this 
run ./configure --enable-cpumining
before make,
GL

Mig75seo, 2021-01-09:
It's works, cool. Thank you

DylanCkawalec, 2021-01-09:
No problem.

RL-R, 2021-01-10:
I have same problem. i had to configure with enable cpu too however with this setup i cannot get my GPU to be recognised in ubuntu on 2 different machines

PS if you comment out line 11958 in miner.c you can compile (but i dont know how this would affect things)

robinh007d, 2021-01-15:
here is a script 

`sudo apt-get install build-essential autoconf automake libtool pkg-config libcurl4-gnutls-dev libjansson-dev uthash-dev libncursesw5-dev libudev-dev libusb-1.0-0-dev libevent-dev libmicrohttpd-dev libhidapi-dev
git clone https://github.com/luke-jr/bfgminer.git
cd bfgminer/
./autogen.sh
./configure --enable-opencl --enable-keccak --disable-other-drivers
sed -i '11958d' miner.c
make`

save as getbfgminer.sh to download and install bfgminer

this configuration allows me to mine with my GPU with the current "reference to cpus" bug

toza-mimoza, 2021-01-29:
Did all of this and still get the same error. The line 11958 is a blank line so I suppose I should have commented the if statement the line before. Still the same output in make.