luke-jr/bfgminer #805: error on windows BFGminer for solo mining

sachinverma1, 2021-02-10:
[image]

hi,

i am trying to do solo mining on my windows .  I am running litecoin Daemon on my windows 10 but i dont know what will be the accurate batch file that can run solo for scrypt algorithm.

i have tried various batch files . sharing you guys a screenshot of error.

i have used these batch files which all are giving me same error.

1. bfgminer -o http://127.0.0.1:9332 -u sachinrpc -p ltc1q0t9vtrc5jw2elwnefwwrz4jjevghrurcfr6ez5
pause

2. bfgminer -a scrypt -o  http://127.0.0.1:9332  -u rpcuser -p rpcpassword --coinbase-addr litecoinaddress  -d listofcards  --no-getwork

3. setx GPU_MAX_ALLOC_PERCENT 100
setx GPU_USE_SYNC_OBJECTS 1

bfgminerr --scrypt --sharelog share.log -o http://127.0.0.1:9332 -u sachinrpc -p ltc1q0t9vtrc5jw2elwnefwwrz4jjevghrurcfr6ez5 --thread-concurrency 8192 --no-gbt 
pause

kindly help me what will be the correct batch file which can work for solo mining.

thank you

wargo32, 2021-03-15:
Compile application or download compiled version.

sachinverma1, 2021-03-15:
Where I can get compiled version of bfgminer.

wargo32, 2021-03-15:
http://bfgminer.org

sachinverma1, 2021-03-15:
I downloaded from this website only... But still it gave me same error.. Windows 64 version I downloaded

wargo32, 2021-03-15:
What do you have in the downloaded directory?

sachinverma1, 2021-03-15:
These are the files which comes after downloading and extracting
[image]
[image]

wargo32, 2021-03-15:
Try type "bfgminer.exe" instead of "bfgminer"

sachinverma1, 2021-03-16:
I tried with bfgminer.exe but it didn't worked either.. Now what should I do to run this miner? 
[image]

wargo32, 2021-03-16:
Try `-k scrypt`

sachinverma1, 2021-03-17:
It is not working with 
bfgminer.exe   - - help
And
bfgminer   - - help

Forget about    - k scrypt

MypowerHD, 2021-05-06:
[image]
same here