luke-jr/bfgminer #806: Error "Pool 0: Unknown stratum msg: {"error":null,"id":auth,"result":true}"

mooko-dev, 2021-02-18:
Hello!
I'm trying to connect to the AntPool. I have created a subaccount with the name "miner888".
I use the console command "bfgminer -o stratum+tcp://ss.antpool.com:3333 -u miner888.worker1 -p anything".
I get the error "Pool 0: Unknown stratum msg: {"error":null,"id":auth,"result":true}".
Also, I tried to connect to the pool using CGMiner with the same parameters and it connects successful.
Why BFGMiner can't connect to the pool?
Version of BFGMiner I have tested is 5.5.0.
Hope for your help!
Thanks in advance!

gaoxuezhao, 2021-03-29:
I have the same problem

petrkr, 2022-01-05:
"id": auth is NOT valid JSON reply, because auth is not in " ", so it's not string...  If it works in CGMiner they have some workaround... This is BUG report to Antminer's pool, because string is NOT valid JSON string.

```
$ echo '{"error":null,"id":auth,"result":true}' | jq . 
parse error: Invalid numeric literal at line 1, column 24
```

versus (notice added " " to auth string)

```
$ echo '{"error":null,"id":"auth","result":true}' | jq . 
{
  "error": null,
  "id": "auth",
  "result": true
}
```

130666ourgame, 2022-01-05:
不是我的问题，我没有向您们提这个这个问题

petrkr, 2022-01-05:
Well..

It's half/half problem on both side. From name "ID" should be number, not string.. But on other hand, antpool should reply string and not unquoted string (which is NOT valid JSON).

For now solution is send number instead string

fixed by https://github.com/petrkr/bfgminer/commit/635041d85e8ec2d5a88cf690a4cc20ee8d98b295