luke-jr/bfgminer #815: Errors while trying solo mining; No servers could be used! Exiting.

gregchi, 2021-04-25:
Hi, I managed to install latest bfgminer from github on my old PC with debian 'Jessie' and the laptop with debian 'Buster'. On both it has the same errors when trying to connect to localhost. I have no idea how to fix that. Tried different versions of bitcoin core 17-21, no positive results. Here is the log:

$: bfgminer --verbose -D -P -T -S auto -o http://127.0.0.1:8332 -u test -p test

 [2021-04-25 06:11:18] Global quota greatest common denominator set to 1
 [2021-04-25 06:11:18] setrlimit: Soft fd limit not being changed from 1024 (FD_SETSIZE=1024; hard limit=1048576)
 [2021-04-25 06:11:18] Started bfgminer 5.5.0-28-g8e0278ad
 [2021-04-25 06:11:18] Loaded configuration file /home/gregi/.bfgminer/bfgminer.conf
 [2021-04-25 06:11:18] usb_devinfo_scan: Error opening device usb:002:002: LIBUSB_ERROR_ACCESS
 [2021-04-25 06:11:18] usb_devinfo_scan: Error opening device usb:002:001: LIBUSB_ERROR_ACCESS
 [2021-04-25 06:11:18] usb_devinfo_scan: Error opening device usb:001:002: LIBUSB_ERROR_ACCESS
 [2021-04-25 06:11:18] usb_devinfo_scan: Error opening device usb:001:001: LIBUSB_ERROR_ACCESS
 [2021-04-25 06:11:18] lowlevel_scan: Found usb device at usb:001:001 (path=(null), vid=1d6b, pid=0002, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found usb device at usb:001:002 (path=(null), vid=0bda, pid=b006, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found usb device at usb:002:001 (path=(null), vid=1d6b, pid=0003, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found usb device at usb:002:002 (path=(null), vid=8086, pid=0a66, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:1f.4 (path=0000:00:1f.4, vid=8086, pid=a123, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:14.0 (path=0000:00:14.0, vid=8086, pid=a12f, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:02.0 (path=0000:00:02.0, vid=8086, pid=191b, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:14.2 (path=0000:00:14.2, vid=8086, pid=a131, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:1c.6 (path=0000:00:1c.6, vid=8086, pid=a116, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:00.0 (path=0000:00:00.0, vid=8086, pid=1910, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:1f.3 (path=0000:00:1f.3, vid=8086, pid=a170, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:01:00.0 (path=0000:01:00.0, vid=10ec, pid=522a, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:16.0 (path=0000:00:16.0, vid=8086, pid=a13a, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:1c.5 (path=0000:00:1c.5, vid=8086, pid=a115, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:04.0 (path=0000:00:04.0, vid=8086, pid=1903, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:02:00.0 (path=0000:02:00.0, vid=10ec, pid=b723, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:1f.0 (path=0000:00:1f.0, vid=8086, pid=a14e, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:1c.0 (path=0000:00:1c.0, vid=8086, pid=a114, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:1f.2 (path=0000:00:1f.2, vid=8086, pid=a121, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:00:17.0 (path=0000:00:17.0, vid=8086, pid=a103, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] lowlevel_scan: Found pci device at pci:0000:03:00.0 (path=0000:03:00.0, vid=10ec, pid=8136, manuf=(null), prod=(null), serial=(null))
 [2021-04-25 06:11:18] INFO: bitfury_detect
 [2021-04-25 06:11:18] /dev/mem trouble: Permission denied
 [2021-04-25 06:11:18] Local bitcoin RPC server on 127.0.0.1:8332 found in /home/username/.bitcoin/bitcoin.conf
 [2021-04-25 06:11:18] Global quota greatest common denominator set to 1
 [2021-04-25 06:11:18] Global quota greatest common denominator set to 1
 [2021-04-25 06:11:18] Added local bitcoin RPC server on 127.0.0.1:8332 as pool 2
 [2021-04-25 06:11:18] Testing pool http://127.0.0.1:8332/#getcbaddr#allblocks
 [2021-04-25 06:11:18] Probing for an alive pool
 [2021-04-25 06:11:18] Testing pool http://127.0.0.1:8332
 [2021-04-25 06:11:18] Testing pool http://127.0.0.1:8332
 [2021-04-25 06:11:18] JSON protocol request:
{"id": 0, "method": "getblocktemplate", "params": [{"capabilities": ["coinbasetxn", "coinbasevalue", "workid", "longpoll", "coinbase/append", "time/increment", "version/force", "version/reduce", "submit/coinbase", "submit/truncate"], "rules": ["csv", "segwit"], "maxversion": 1073741823}]}
 [2021-04-25 06:11:18] JSON protocol request:
{"id": 0, "method": "getblocktemplate", "params": [{"capabilities": ["coinbasetxn", "coinbasevalue", "workid", "longpoll", "coinbase/append", "time/increment", "version/force", "version/reduce", "submit/coinbase", "submit/truncate"], "rules": ["csv", "segwit"], "maxversion": 1073741823}]}
 [2021-04-25 06:11:18] JSON protocol request:
{"id": 0, "method": "getblocktemplate", "params": [{"capabilities": ["coinbasetxn", "coinbasevalue", "workid", "longpoll", "coinbase/append", "time/increment", "version/force", "version/reduce", "submit/coinbase", "submit/truncate"], "rules": ["csv", "segwit"], "maxversion": 1073741823}]}
 [2021-04-25 06:11:18] Pool 2: Expire in 0 ms for 6 (transfer 0x7f6874000b20)
 [2021-04-25 06:11:18] Pool 0: Expire in 0 ms for 6 (transfer 0x7f6864000b20)
 [2021-04-25 06:11:18] Pool 1: Expire in 0 ms for 6 (transfer 0x7f686c000b20)
 [2021-04-25 06:11:18] Pool 2: Expire in 60000 ms for 8 (transfer 0x7f6874000b20)
 [2021-04-25 06:11:18] Pool 0: Expire in 60000 ms for 8 (transfer 0x7f6864000b20)
 [2021-04-25 06:11:18] Pool 1: Expire in 60000 ms for 8 (transfer 0x7f686c000b20)
 [2021-04-25 06:11:18] Pool 2:   Trying 127.0.0.1...
 [2021-04-25 06:11:18] Pool 0:   Trying 127.0.0.1...
 [2021-04-25 06:11:18] Pool 2: TCP_NODELAY set
 [2021-04-25 06:11:18] Pool 0: TCP_NODELAY set
 [2021-04-25 06:11:18] Pool 1:   Trying 127.0.0.1...
 [2021-04-25 06:11:18] Pool 1: TCP_NODELAY set
 [2021-04-25 06:11:18] Pool 2: Expire in 200 ms for 4 (transfer 0x7f6874000b20)
 [2021-04-25 06:11:18] Pool 1: Expire in 200 ms for 4 (transfer 0x7f686c000b20)
 [2021-04-25 06:11:18] Pool 0: Expire in 200 ms for 4 (transfer 0x7f6864000b20)
 [2021-04-25 06:11:18] Pool 1: Connected to 127.0.0.1 (127.0.0.1) port 8332 (#0)
 [2021-04-25 06:11:18] Pool 2: Connected to 127.0.0.1 (127.0.0.1) port 8332 (#0)
 [2021-04-25 06:11:18] Pool 1: Server auth using Basic with user 'test'
 [2021-04-25 06:11:18] Pool 2: Server auth using Basic with user 'test'
 [2021-04-25 06:11:18] Pool 0: Connected to 127.0.0.1 (127.0.0.1) port 8332 (#0)
 [2021-04-25 06:11:18] Pool 0: Server auth using Basic with user 'test'
[... 488 lines cut ...]

 [2021-04-25 06:11:19] Pool: http://127.0.0.1:8332/#getcbaddr#allblocks
 [2021-04-25 06:11:19]  Share submissions: 0
 [2021-04-25 06:11:19]  Accepted shares: 0
 [2021-04-25 06:11:19]  Rejected shares: 0 + 0 stale (-nan%)
 [2021-04-25 06:11:19]  Accepted difficulty shares: 0
 [2021-04-25 06:11:19]  Rejected difficulty shares: 0
 [2021-04-25 06:11:19]  Network transfer:  2.98 /  0.00 MB  ( 2.98 /  0.00 MB/s)
 [2021-04-25 06:11:19]  Efficiency (accepted * difficulty / 2 KB): 0.00
 [2021-04-25 06:11:19]  Items worked on: 0
 [2021-04-25 06:11:19]  Unable to get work from server occasions: 0
 [2021-04-25 06:11:19]  Submitting work remotely delay occasions: 0

 [2021-04-25 06:11:19] Summary of per device statistics:

 [2021-04-25 06:11:19] CPU0        | 20s:  0.0 avg:  0.0 u:  0.0  h/s | A:0 R:0+0(none) HW:0/none
 [2021-04-25 06:11:19] CPU1        | 20s:  0.0 avg:  0.0 u:  0.0  h/s | A:0 R:0+0(none) HW:0/none
[... cut ...]
 [2021-04-25 06:11:19] CPU3        | 20s:  0.0 avg:  0.0 u:  0.0  h/s | A:0 R:0+0(none) HW:0/none
 [2021-04-25 06:11:19] CPU4        | 20s:  0.0 avg:  0.0 u:  0.0  h/s | A:0 R:0+0(none) HW:0/none
 [2021-04-25 06:11:19] CPU5        | 20s:  0.0 avg:  0.0 u:  0.0  h/s | A:0 R:0+0(none) HW:0/none
 [2021-04-25 06:11:19] CPU6        | 20s:  0.0 avg:  0.0 u:  0.0  h/s | A:0 R:0+0(none) HW:0/none
 [2021-04-25 06:11:19] CPU7        | 20s:  0.0 avg:  0.0 u:  0.0  h/s | A:0 R:0+0(none) HW:0/none
 [2021-04-25 06:11:19]  
**_No servers could be used! Exiting._**

ikeedge, 2021-04-25:
Bro you’re probably better with this shit than me. I . got four old wallets
through black chain that I can’t access I dove right in and I’m still
figuring shit out

gregchi, 2021-04-25:
On debian jessie it started working when I added --coinbase-address but it only works with bitcoind ver 17x.I tried higher versions and same errors occure. Happy to say it uses old radeon HD 5770 GPU and old Phenom X2 CPU :)  On laptop it works with latest bitcore 0.21 ver. - I would assume some libraries on debian jessie will must be upgraded to make it work with bitcoin 0.21...
[image]

[image]

closed 2021-04-25 by gregchi