luke-jr/bfgminer #816: Raspberry Pi 4 Error when "make -j 2"

IDontModLobbies, 2021-05-02:
In file included from driver-gekko.c:1:
driver-gekko.c: In function \u2018compac_mine\u2019:
driver-gekko.h:8:24: warning: implicit declaration of function \u2018pthread_yield\u2019; did you mean \u2018thread_yield\u2019? [-Wimplicit-function-declaration]
 #define thread_yield() pthread_yield(NULL)
                        ^~~~~~~~~~~~~
driver-gekko.c:579:4: note: in expansion of macro \u2018thread_yield\u2019
    thread_yield();
    ^~~~~~~~~~~~
  CCLD     cgminer
/usr/bin/ld: cgminer-util.o: in function `parse_method.part.18':
util.c:(.text+0x8c80): undefined reference to `json_string_length'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:891: cgminer] Error 1
make[2]: Leaving directory '/home/pi/git/vthoang/cgminer'
make[1]: *** [Makefile:1835: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/git/vthoang/cgminer'
make: *** [Makefile:794: all] Error 2

IDontModLobbies, 2021-05-02:
Followed this guide and I get errors upon installation.
[NewPac BM1387 USB Stick Miner User Guide](https://www.manualslib.com/manual/1608752/Gekkoscience-Newpac-Bm1387.html?page=7#manual)

`sudo apt-get update
sudo apt-get upgrade -y

sudo apt-get install -y build-essential git libusb-1.0-0-dev libusb-1.0-0 libcurl4-openssl-dev libncurses5-dev libudev-dev screenlibtool automake pkg-config libjansson-dev

mkdir -p git/vthoang; cd git/vthoang
git clone https://github.com/vthoang/cgminer.git
cd cgminer

CFLAGS="-O2" ./autogen.sh --enable-gekko
make -j 2`

C-O-S, 2022-04-13:
Those instructions are for cgminer, not bfgminer.  

That said, try the instructions here:
https://bitcointalk.org/index.php?topic=5355470.msg58200639#msg58200639

It should help you compile a working gekko driver you can use.

IDontModLobbies, 2022-04-13:
> Those instructions are for cgminer, not bfgminer.  
> 
> 
> 
> That said, try the instructions here:
> 
> https://bitcointalk.org/index.php?topic=5355470.msg58200639#msg58200639
> 
> 
> 
> It should help you compile a working gekko driver you can use.
> 
> 

I'm trying to make it work for an ASIC BlockEruptor. I will definitely try this out when I get home from work. Fingers crossed!

C-O-S, 2022-04-14:
If you're only using a BlockErupter, then you may need to replace this line:
CFLAGS="-O2 -march=native -fcommon" ./autogen.sh --enable-gekko --enable-icarus

With this one:
CFLAGS="-O2 -march=native -fcommon" ./autogen.sh  --enable-blockerupter

Per the README:
https://github.com/kanoi/cgminer/blob/master/README

To ensure you enable the driver as part of the build for the executable.

Best of luck!

IDontModLobbies, 2022-04-14:
Everything is working now, except the device cannot be found. Are these considered blockeruptors? 

[image]

C-O-S, 2022-04-14:
Try this:

sudo cgminer --ndevs

If it lists the actual device(s), then your issue is related to your OS not granting your user account access to those USB devices.  

You can shortcut troubleshooting by trying a test run of the devices with sudo prepended to the command like the above, but realistically that's not a long-term great solution.

IDontModLobbies, 2022-04-15:
> sudo cgminer --ndevs

Sadly, I get this error. 

`sudo: cgminer: command not found`

When I go to USB Management, it does state that there is one device. However, it does not start the mining process.

Saikatsaha1996, 2022-05-05:
> > sudo cgminer --ndevs
> 
> Sadly, I get this error.
> 
> `sudo: cgminer: command not found`
> 
> When I go to USB Management, it does state that there is one device. However, it does not start the mining process.

cd cgminer
sudo cgminer --ndevs