luke-jr/bfgminer #833: Bfgminer Clone from github

Derk2020, 2022-04-17:
Hi,
I have the Problem that I can't use the ./autogen.sh
When I do this there is a Problem with Clone from github.
File could not be found

Can anyone help?

kazmek, 2022-05-11:
I was trying to compile it myself too. I received a url in the errors it threw. [https://github.blog/2021-09-01-improving-git-protocol-security-github/](url)
I've been out of the loop for a bit. I'll poke around a bit more. I'm thinking the urls just need to be updated to HTTPS since I see its still trying to clone stuff into the dir using unauthenticated protocols.

MarkSierra, 2022-05-24:
I have the same issue. A couple of months ago it worked just fine.

MarkSierra, 2022-06-04:
I found a fix!

After "git clone https://github.com/luke-jr/bfgminer.git" type in "cd bfgminer".
Then type in "sudo nano .gitmodules". In this file you have to replace every "git://" with "https://".
Exit and save the file.

Then execute the "sudo ./autogen.sh" command and it should work :)

blindrun, 2022-08-09:
> I found a fix!
> 
> After "git clone https://github.com/luke-jr/bfgminer.git" type in "cd bfgminer". Then type in "sudo nano .gitmodules". In this file you have to replace every "git://" with "https://". Exit and save the file.
> 
> Then execute the "sudo ./autogen.sh" command and it should work :)

you need to change the git:// to https:// in the ~bfgminer/.git/configure file as well

nymbuzcumulus, 2024-08-28:
Dear Friend,,,
I have installed the yiimp server it seems its working, I have the console open on a second pc on the same network,,, for me thats a triumph since I am not the most qualified programmer. I humbly recognize my limits..
I want to ask you if you would be interested on helping me to do the right settings to Yiimp.. I have bitcoin-qt runing on the network as well.

The goal> We have here 50 asics.. between s9 + s19 +whatsminers,, in between all we do 3.5PH !
We are meant to push the solo blocks mode with all that power and seek for the block reward.

My Proposal is help us to do the whole thing work and we go half reward with you on the first block found... what you think ?
Would love to have your Mastery participation in our project,,,

You canb contact me via:
email: [nymbuz13@gmail.com](mailto:nymbuz13@gmail.com)
whatsapp: +525564907240
Sally Sheridan