luke-jr/bfgminer #844: Bfgminer for litecoin

sachinverma1, 2022-08-12:
Hi, .

I am trying to mine litecoin solo...  I want to know does bfgminer works for litecoin...

If yes then any article will be helpful for me.

Thank you

wareck, 2023-01-02:
yes...
enable scrypt during configuration (and you also need hardware support scrypt/litecoin)
I'm using bfgminer since a while for liecoin...

sachinverma1, 2023-01-02:
Thank you for your response.. How I can enable scrypt in bfgminer while
mining on litecoin...?

wareck, 2023-01-02:
download bfgminer from source , autogen config files, then configure

sudo apt-get update -y
sudo apt-get install git build-essential autoconf automake libtool pkg-config libcurl4-gnutls-dev \
    libjansson-dev uthash-dev libncursesw5-dev libudev-dev libusb-1.0-0-dev \
    libevent-dev libmicrohttpd-dev libhidapi-dev libgcrypt20-dev -y

git clone https://github.com/luke-jr/bfgminer.git
cd bfgminer
./autogen.sh
./configure --enable-scrypt
make

If you want to use litecoin hardware, you can add your driver like zeusminer:

./configure --enable-scrypt --enable-zeus
make

sachinverma1, 2023-01-02:
Cool... Can I do it on windows 10?

sachinverma1, 2023-01-06:
W to enable Nvidia GPU support in bfgminer...

luke-jr, 2024-09-09:
Scamcoins are not supported

closed 2024-09-09 by luke-jr

sachinverma1, 2024-09-09:
Issue solved