luke-jr/bfgminer #845: Error during "./autogen" step of build on Linux:   "fatal: unable to connect to git.ozlabs.org:"

amritabithi, 2022-08-31:
It seems that this respository for ccan-upstream is offline, here is my error below:

```
Cloning into '/home/bfgminer/ccan-upstream'...
fatal: unable to connect to git.ozlabs.org:
git.ozlabs.org[0: 150.107.74.76]: errno=Connection timed out
git.ozlabs.org[1: 2404:9400:2221:ea00::2]: errno=Network is unreachable
```

amritabithi, 2022-10-15:
I was able to build by changing the git:// URLs to https:// in the file **/bfgminer/.git/config** , I have pasted my version which is working for me below:

```
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
[remote "origin"]
	url = https://github.com/luke-jr/bfgminer.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "bfgminer"]
	remote = origin
	merge = refs/heads/bfgminer
[submodule "ccan"]
	active = true
	url = https://git.ozlabs.org/~ccan/ccan
[submodule "knc-asic"]
	active = true
	url = https://github.com/KnCMiner/knc-asic
[submodule "libbase58"]
	active = true
	url = https://github.com/luke-jr/libbase58.git
[submodule "libblkmaker"]
	active = true
	url = https://github.com/bitcoin/libblkmaker.git
```

JAG0991, 2023-02-18:
> I was able to build by changing the git:// URLs to https:// in the file **/bfgminer/.git/config** , I have pasted my version which is working for me below:
>
> ```
> [core]
> 	repositoryformatversion = 0 [...]

Also had this issue. This fixed it. Thank you!

yhilgjff, 2024-03-17:
Thank you, that also fixed it for me.