luke-jr/bfgminer #853: Does this repository contain an Trojan?

erdincay, 2023-12-23:
I am curious to hear what the reason is, that the source code contained here, has been detected as Trojan:

AhnLab-V3: Malware/PHP.Generic.SC155886
Antiy-AVL: Trojan/Script.Agent
Arcabit: Application.PHP.Miner.C
Avast: BV:Miner-CM [Trj]
AVG: BV:Miner-CM [Trj]
BitDefender: Application.PHP.Miner.C
DrWeb: Tool.BtcMine.83
Emsisoft: Application.PHP.Miner.C (B)
GData: Application.PHP.Miner.C
Gridinsoft (no cloud): Malware.U.GenericMC.cc
Lionic: Trojan.ZIP.Generic.4!c
MAX: Malware (ai Score=86)
Skyhigh (SWG): Artemis
Trellix (FireEye): Application.PHP.Miner.C
VIPRE: Application.PHP.Miner.C
Xcitium: Application.Script.BitMiner.A@7xk8f5

Thanks for your efforts anyhow.

Best Regards.

olic110, 2023-12-23:
Try anti virus www.avg.comAfter full scan of computing software and hardware you may find at first the same problems however After scanning you tek and software you will findthee plenty room with avg to adapt with crypto. Thank you Mad oli /the internet   🛸👽Sent from my iPhone

ryao, 2024-01-06:
The FAQ in the README says:

```
Q: Is this a virus?
A: As BFGMiner is being packaged with other trojan scripts, some antivirus
software is falsely accusing bfgminer.exe as being the actual virus, rather than
whatever it is being packaged with. If you installed BFGMiner yourself from a
reputable source then you do not have a virus on your computer. Complain to your
antivirus software company. They seem to be flagging even source code from
BFGMiner as malicious now, even though text source files can't do anything by
themselves.
```

closed 2024-09-09 by luke-jr