luke-jr/bfgminer #857: @luke miners not found as in the network

nymbuzcumulus, 2024-08-30:
Dear Luke.. 
i recon u are The Genius! 
Dear need your precious help here..
Bfg Miner is finally working with Kali.. *(yup it's crazy) is performing beautiful with the 5 cores of the desktop where is installed, is connecting to bitcoin core, and showing several hashes... I love it ! 

By misfortune , and after reading 300 thousand blogs, I just cant make mi miners on the very same network to make a hit.. I really don't know much , and humbly I am not the most advanced programmer. 

Please Please need your super help . I am stuck and ignore what is the proper config to make miners to hit.. I have 3.5 PH power and we need to make them value.. will add my configs.

nymbuzcumulus, 2024-08-30:
this is my .bitcoin conf
### IPv4/IPv6 mode ###
# This mode requires uPnP feature on your router to allow Bitcoin Core to accept incoming connections.

upnp=0

### Tor mode ###
# This mode requires tor (https://www.torproject.org/download/) to be running at the proxy address below.
# No configuration is needed on your router to allow Bitcoin Core to accept incoming connections.
#proxy=127.0.0.1:9050
#bind=127.0.0.1
#onlynet=onion
#enable JSON/RPC API
#Server IP address
#IP addresses of clients to make calls:
#All computers with addres 192.168.0.xxx

server=1
daemon=1
listen=1
gen=0
rpcuser=nymbuz
rpcpassword=BTC
rcpallowip=12.34.56.789
rpcallowip=127.0.0.1
rpcport=8333
rpcport = 8332
rpcport=4028
maxconnections=120
dbcache=14096
par=1
checkblocks=24
checklevel=0
rpccookiefile=/home/nymbuz/bitcoin-core/.bitcoin/.cookie
rpcbind=127.0.0.1
rpcallowip=127.0.0.1
mintxfee=0.0000001
walletrbf=1
upgradewallet=1
walletbroadcast=1
flushwallet=1
privdb=1
rpcallowip=192.168.0.0/24
rpcbind=127.0.0.1:8332
rpcbind=127.0.0.1:8333
rpcallowip=0.0.0.0/0
rpcbind=0.0.0.0
zmqpubrawblock=tcp://0.0.0.0:8332
zmqpubrawtx=tcp://0.0.0.0:8333

nymbuzcumulus, 2024-08-30:
This is my bfgminer conf 
{
"pools" : [
	{
		"url" : "http://127.0.0.1:8333/#getcbaddr#allblocks",
		"pool-proxy" : "socks5://127.0.0.1:1080"
		"user" : "nymbuz",
		"pass" : "BTC",
		"coinbase-addr 1PSPAV56s2N2APRcy2XbVSKbhL3fZurWrX"
		"pool-priority" : "0"
	}
]
,
"algo" : "c",
"api-mcast-port" : "8333",
"api-port" : "4028",
"expiry" : "120",
"expiry-lp" : "3600",
"failover-switch-delay" : "300",
"log" : "20",
"no-pool-disable" : true,
"no-client-reconnect" : true,
"no-show-processors" : true,
"no-show-procs" : true,
"no-unicode" : true,
"queue" : "5",
"quiet-work-updates" : true,
"quiet-work-update" : true,
"scan-time" : "60",
"skip-security-checks" : "0",
"submit-stale" : true,
"temp-hysteresis" : "3",
"shares" : 0,
"kernel-path" : "/usr/local/share/bfgminer"
}
"scan" : [
	"/dev/ttyUSB0",
	"/dev/ttyUSB1",
	"/dev/ttyUSB2",
	"/dev/ttyUSB3",
	"bitmain:auto",
	"antminer:all",
	"bitmain:auto",
	"antminer:all",
	"all"
],
{
"stratum-port" : 8332,
"api-allow" : "W:192.168.0.0/24"
"api-allow" : "w:192.168.0.173:8332"
"api-allow" : "w:192.168.0.173:4028"
}
}

nymbuzcumulus, 2024-08-30:
The tests I am doing on an s9 miner point: 

stratum+tcp:192.168.0.139:8332 
Same tried on ports 8333, 3333, 4028... I just dont make a hit. 

The other issue, I believe that I might be missing a stratum config... might that be ?

nymbuzcumulus, 2024-08-30:
[image]
thats a screenshot of bfg miner

nymbuzcumulus, 2024-08-30:
there was a moment on some config that I was able to see Proxy conection sick, and the ip of the antminer coming up... but I lost that config tryiing to make it healthy

nymbuzcumulus, 2024-08-30:
on the Bitcoin core, I see on the Node window Peers, that traffic os working propperly  here a shot...

[image]

nymbuzcumulus, 2024-08-30:
[image]
I believe my fail is at this pont where pool 1 is dead //// not sure

nymbuzcumulus, 2024-08-30:
the new debug. after this screenshot, comes with a legend that caught my eye: Stratimfails to initiate.... Voila  I think thats the issue ... now .. how do I fix that?

nymbuzcumulus, 2024-09-09:
Dear friends ,
Somehow  we have been able to launch and run the Bfgminer 5.5.0-34....but we have some small issues and need your accurate and good timing help.

Issue #1
Bitminers s9, s19 and whats miners fail to work with bfgminer if ASIC BOOST is enabled.
Disabling ASIC BOOST makes the pool ALIVE on miners console, but only works 5 minutes, overheat the hashboards and kills them xxxxx-xxxxx-xxxxx- and so.

Issue #2
As soon as same miners start to hash ,, (they are 14th and 13th asics) The BFGMINER shows them on PXY0 and shows them hashing only 140GS max rate , then dies.

In highest hopes you find this email and can help to support these 2 issues...

Greatest Regards
Nymbuz

RicksGhost, 2024-11-14:
I need help badly..I'm not a techie and don't know much..I know I've downloaded Bitcoin core to my iOS and used fully noded but it says network not found.and I need help please..I had a dev that was supposed to be helping me but turns out he was helping himself..very depressed over this and don't know how to fix it...[image]

nymbuzcumulus, 2024-11-14:
Don’t waste your time ! This is a crap it will never work!

RicksGhost, 2024-11-14:
Wow what do I do now?

nymbuzcumulus, 2024-11-14:
I made a solo pool, for bch and BTC in the next hrs will be migrating to a large server- give a try nymbuzpool.org bch is fabulous 40% more than BTCFind me on telegram as nymbuz

nymbuzcumulus, 2024-11-14:
Contact me on ***@***.***_Cumulus

nymbuzcumulus, 2024-11-15:
Server is up and running check nymbuzpool org  or contact telegram Nymbuz_Cumulus let me help u ! I wasted 4 months with this until Mr Luke instead of helping claims defectuos my miners! Was not nice at all!!