luke-jr/bfgminer #858: ASIC BOOST

nymbuzcumulus, 2024-09-09:
@luke-jr 
Dear friends ,
Somehow we have been able to launch and run the Bfgminer 5.5.0-34....but we have some small issues and need your accurate and good timing help.

Issue https://github.com/luke-jr/bfgminer/issues/1
Bitminers s9, s19 and whats miners fail to work with bfgminer if ASIC BOOST is enabled.
Disabling ASIC BOOST makes the pool ALIVE on miners console, but only works 5 minutes, overheat the hashboards and kills them xxxxx-xxxxx-xxxxx- and so.

Issue https://github.com/luke-jr/bfgminer/issues/2
As soon as same miners start to hash ,, (they are 14th and 13th asics) The BFGMINER shows them on PXY0 and shows them hashing only 140GS max rate , then dies.

In highest hopes you find this email and can help to support these 2 issues...

Greatest Regards
Nymbuz

nymbuzcumulus, 2024-09-09:
[image]

nymbuzcumulus, 2024-09-09:
There it is an s9 working ,, thats a fact ! 
But had to turn off ASIC BOOST in order that the miner  finds an ALIVE POOL ... it dies 5 minutes later !

luke-jr, 2024-09-09:
ASICBoost is a violation of the Bitcoin protocol and has never been supported by BFGMiner.

If your machines are defective, there's nothing I can do about that...

nymbuzcumulus, 2024-09-09:
Dear Look... Vnish, Braiins, and every modern asic includes the asic boost as a tool. 
Every ASIC from s9, s19, s21 comes with it embedded , and new firmwares also do. 

It is really awefull that after many hrs of trying to install your dev comes out that ASIC Miners do not work... Please fix that ASIC BOOST thingy.. YOu MUst !